mkeeter/antimony #237: Git master build fails on the lemon step

prokoudine, 2023-01-20:
Ubuntu 22.04, lemon 3.37.2-2ubuntu0.1, all steps in BUILDING.md done

```
[17/131] Generating v2syntax.lemon.hpp, v2syntax.lemon.cpp
FAILED: lib/fab/v2syntax.lemon.hpp lib/fab/v2syntax.lemon.cpp /home/alex/soft/cad/antimony/build/lib/fab/v2syntax.lemon.hpp /home/alex/soft/cad/antimony/build/lib/fab/v2syntax.lemon.cpp 
cd /home/alex/soft/cad/antimony/build/lib/fab && /usr/local/bin/lemon -q -c -s /home/alex/soft/cad/antimony/lib/fab/src/tree/v2syntax.y && mv /home/alex/soft/cad/antimony/lib/fab/src/tree/v2syntax.c /home/alex/soft/cad/antimony/build/lib/fab/v2syntax.lemon.cpp && mv /home/alex/soft/cad/antimony/lib/fab/src/tree/v2syntax.h /home/alex/soft/cad/antimony/build/lib/fab/v2syntax.lemon.hpp
Can't open the template file "lempar.c".
Parser statistics: 60 terminals, 8 nonterminals, 58 rules
                   159 states, 0 parser table entries, 0 conflicts
[18/131] Building CXX object lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /home/alex/soft/cad/antimony/lib/graph/inc/graph/proxy.h:3,
                 from /home/alex/soft/cad/antimony/lib/graph/tests/datum.cpp:8:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
[19/131] Building CXX object lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /home/alex/soft/cad/antimony/lib/graph/inc/graph/proxy.h:3,
                 from /home/alex/soft/cad/antimony/lib/graph/tests/link.cpp:8:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
[21/131] Building CXX object lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /home/alex/soft/cad/antimony/lib/graph/inc/graph/proxy.h:3,
                 from /home/alex/soft/cad/antimony/lib/graph/tests/script.cpp:8:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
[22/131] Building CXX object lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /home/alex/soft/cad/antimony/lib/graph/inc/graph/proxy.h:3,
                 from /home/alex/soft/cad/antimony/lib/graph/tests/main.cpp:6:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

mkeeter, 2023-01-20:
If you revert back one commit (to 8b805c674adad536f9dd552b4be75fadcb3c7db6), does it work?

It seems like the failing step is
```
/usr/local/bin/lemon -q -c -s /home/alex/soft/cad/antimony/lib/fab/src/tree/v2syntax.y
```

Can you try running that on its own, possibly using `ptrace` to figure out where it's trying to find `lempar.c`?  It appears that `lempar.c` is a part of `lemon`, but dunno where Ubuntu normally puts it.

prokoudine, 2023-01-20:
> If you revert back one commit (to https://github.com/mkeeter/antimony/commit/8b805c674adad536f9dd552b4be75fadcb3c7db6), does it work?

It now fails in a different place:

```
[22/131] Building CXX object lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /home/alex/soft/cad/antimony/lib/graph/inc/graph/proxy.h:3,
                 from /home/alex/soft/cad/antimony/lib/graph/tests/main.cpp:6:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

I'll see about ptracing that command.

prokoudine, 2023-01-20:
```
strace /usr/local/bin/lemon -q -c -s /home/alex/soft/cad/antimony/lib/fab/src/tree/v2syntax.y
execve("/usr/local/bin/lemon", ["/usr/local/bin/lemon", "-q", "-c", "-s", "/home/alex/soft/cad/antimony/lib"...], 0x7ffeb96d4190 /* 58 vars */) = 0
brk(NULL)                               = 0x562af5093000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffcb9209d40) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f845e2cd000
readlink("/proc/self/exe", "/usr/local/bin/lemon", 4096) = 20
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/glibc-hwcaps/x86-64-v3/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/glibc-hwcaps/x86-64-v3", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/glibc-hwcaps/x86-64-v2/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/glibc-hwcaps/x86-64-v2", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/tls/haswell/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/tls/haswell/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/tls/haswell/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/tls/haswell", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/tls/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/tls", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/haswell/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/haswell/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/haswell/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/haswell", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/lib", {st_mode=S_IFDIR|0755, st_size=12288, ...}, 0) = 0
openat(AT_FDCWD, "/usr/local/bin/../lib/glibc-hwcaps/x86-64-v3/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/glibc-hwcaps/x86-64-v3", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/glibc-hwcaps/x86-64-v2/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/glibc-hwcaps/x86-64-v2", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/tls/haswell/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/tls/haswell/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/tls/haswell/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/tls/haswell", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/tls/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/tls", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/haswell/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/haswell/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/haswell/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/haswell", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib/x86_64", 0x7ffcb9208f60, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/bin/../lib/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/local/bin/../lib", {st_mode=S_IFDIR|0755, st_size=12288, ...}, 0) = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=210767, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 210767, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f845e299000
[... cut ...]
read(3, "#define TOKEN_PLUS              "..., 4096) = 2891
read(3, "", 4096)                       = 0
close(3)                                = 0
newfstatat(1, "", {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x8), ...}, AT_EMPTY_PATH) = 0
write(1, "Parser statistics: 60 terminals,"..., 58Parser statistics: 60 terminals, 8 nonterminals, 58 rules
) = 58
write(1, "                   159 states, 0"..., 67                   159 states, 0 parser table entries, 0 conflicts
) = 67
exit_group(1)                           = ?
+++ exited with 1 +++

```

prokoudine, 2023-01-20:
locate tells me this:

/usr/local/share/lemon/lempar.c
/usr/share/lemon/lempar.c

The former one was installed along with brl-cad. The latter one was installed as part of the lemon package on Ubuntu.

mkeeter, 2023-01-20:
Okay, this appears to be a disagreement between the `lemon` executable and the Ubuntu package.

`lemon` looks for `lempar.c` in the same directory as the binary (see `pathsearch` [here](https://www.sqlite.org/src/file?name=tool/lemon.c&ci=trunk)), i.e. `/usr/local/bin/lempar.c`, but Ubuntu has put it in `/usr/share/lemon/lempar.c`.

Homebrew patches this [here](https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/lemon.rb#L30-L31); perhaps Ubuntu should be doing something similar?

(I also can't explain why this worked in the past!)